anestisb/android-prepare-vendor #110: API 27 support (oreo-mr1-release / Android 8.1)

thestinger, 2017-12-05:
Google released Android 8.1 today with a bump to the API level. They moved to a newer snapshot of master as the base across the board but I don't think there's going to be anything requiring special attention in android-prepare-vendor.

thestinger, 2017-12-05:
It looks like an update to oatdump will be required at a minimum:

```
[-] DEX dump from '/home/strcat/projects/copperheados/android-prepare-vendor/angler/opm1.171019.011/factory_imgs_data/system/app/HwMMITest/oat/arm64/HwMMITest.odex' failed
Failed to open oat file from '/home/strcat/projects/copperheados/android-prepare-vendor/angler/opm1.171019.011/factory_imgs_data/system/app/HwMMITest/oat/arm64/HwMMITest.odex': Failed to load vdex file '/home/strcat/projects/copperheados/android-prepare-vendor/angler/opm1.171019.011/factory_imgs_data/system/app/HwMMITest/oat/arm64/HwMMITest.vdex' Vdex file is not valid
```

anestisb, 2017-12-06:
Yes they bumped both the OAT & VDEX file versions so need to recompile the tools. Will try to release a first build later today.

thestinger, 2017-12-06:
These are also gone on marlin / sailfish for API 27:

        "system/lib64/vendor.qti.atcmdfwd@1.0.so",
        "system/lib64/vendor.qti.qcril.am@1.0.so",

anestisb, 2017-12-06:
Damn. Unfortunately Google has not picked the oatdump unquicken [contributions](https://android.googlesource.com/platform/art/+/a1f56a8dddb88f5377a7dd4ec79640103c713d30) despite being developed against the same VDEX version (010). This means that bullhead/angler support will take some more time, since I'll have to manually backport them again in the 8.1.0 oatdump.

thestinger, 2017-12-06:
Unfortunately that probably won't ship until 9.0. I've run into similar issues where I land a change and it's one of the few commits that doesn't make it from the master commits to the release. It hasn't happened much recently because I've had no luck landing anything for a while and pretty much gave up on that...

anestisb, 2017-12-06:
The naked configs generated from a Linux host are good to go for all four devices (Pixel, Pixel XL, 5x, 6p). The full ones & the macOS oatdump bins will be done at some point tomorrow.

anestisb, 2017-12-06:
I haven't flashed and tested the builds yet for the matching devices. Will do later today.

mydongistiny, 2017-12-06:
So far on marlin I've found these changes

Removed:

system/etc/permissions/vzw_sso_permissions.xml
system/etc/permissions/com.android.vzwomatrigger.xml
system/etc/sysconfig/dialer_experience.xml
system/priv-app/HotwordEnrollmentWCD9335
system/priv-app/VZWAPNService/VZWAPNService.apk
system/priv-app/VzwLcSilent/VzwLcSilent.apk

Added:

system/etc/sysconfig/framework-sysconfig.xml
system/priv-app/HotwordEnrollmentOKGoogleWCD9335/HotwordEnrollmentOKGoogleWCD9335.apk
system/priv-app/HotwordEnrollmentTGoogleWCD9335/HotwordEnrollmentTGoogleWCD9335.apk
system/priv-app/HotwordEnrollmentXGoogleWCD9335/HotwordEnrollmentXGoogleWCD9335.apk

mydongistiny, 2017-12-08:
device-vendor-marlin.mk doesn't get generated. It gets generated as device-vendor.mk so it doesn't get picked up by the aosp device tree.

anestisb, 2017-12-09:
Support has been added for all 4 devices. Please report potential issues in a separate thread.

closed 2017-12-09 by anestisb