anestisb/android-prepare-vendor #111: sailfish: investigate vendor build.prop diffs

anestisb, 2017-12-07:
```
$ diff -u <(cat ~/Developer/android/android-prepare-vendor/sailfish/opm1.171019.011/factory_imgs_data/vendor/build.prop | grep -v "^#" | sort) <(cat vendor/build.prop | grep -v "^#" | sort)
--- /dev/fd/63	2017-12-07 20:00:18.309316495 +0200
+++ /dev/fd/62	2017-12-07 20:00:18.305316395 +0200
@@ -1,5 +1,4 @@

-
 aaudio.hw_burst_min_usec=2000
 aaudio.mmap_exclusive_policy=2
 aaudio.mmap_policy=2
@@ -50,22 +49,17 @@
 qdcm.diagonal_matrix_mode=1
 qdcm.only_pcc_for_trans=1
 rild.libpath=/vendor/lib64/libril-qc-qmi-1.so
-ro.adb.secure=1
 ro.atrace.core.services=com.google.android.gms,com.google.android.gms.ui,com.google.android.gms.persistent
 ro.board.platform=msm8996
-ro.boot.vendor.overlay.theme=com.google.android.theme.pixel
 ro.bt.bdaddr_path=/sys/module/bdaddress/parameters/bdaddress
 ro.camera.notify_nfc=1
 ro.carrier=unknown
-ro.com.android.dataroaming=false
-ro.com.android.prov_mobiledata=false
+ro.com.android.dataroaming=true
 ro.com.google.clientidbase=android-google
-ro.com.google.ime.theme_id=5
-ro.config.alarm_alert=Flow.ogg
-ro.config.notification_sound=Chime.ogg
-ro.config.ringtone=Zen.ogg
+ro.config.alarm_alert=Alarm_Classic.ogg
+ro.config.notification_sound=pixiedust.ogg
+ro.config.ringtone=Ring_Synth_04.ogg
 ro.config.vc_call_vol_steps=7
-ro.control_privapp_permissions=enforce
 ro.cp_system_other_odex=1
 ro.crypto.scrypt_params=13:3:1
 ro.error.receiver.system.apps=com.google.android.gms
@@ -74,6 +68,8 @@
 ro.facelock.est_max_time=600
 ro.facelock.rec_timeout=3500
 ro.frp.pst=/dev/block/platform/soc/624000.ufshc/by-name/frp
+ro.hardware.egl=adreno
+ro.hardware.fingerprint=fpc
 ro.hardware.power=marlin
 ro.hwui.drop_shadow_cache_size=6
 ro.hwui.gradient_cache_size=1
@@ -100,18 +96,17 @@
 ro.telephony.default_network=10
 ro.url.legal.android_privacy=http://www.google.com/intl/%s/mobile/android/basic/privacy.html
 ro.url.legal=http://www.google.com/intl/%s/mobile/android/basic/phone-legal.html
-ro.vendor.build.date=Mon Nov 13 02:36:21 UTC 2017
-ro.vendor.build.date.utc=1510540581
-ro.vendor.build.fingerprint=google/sailfish/sailfish:8.1.0/OPM1.171019.011/4448085:user/release-keys
+ro.vendor.build.date=Thu Dec 7 17:56:56 EET 2017
+ro.vendor.build.date.utc=1512662216
+ro.vendor.build.fingerprint=Android/aosp_sailfish/sailfish:8.1.0/OPM1.171019.011/anesti12071756:user/test-keys
 ro.vendor.extension_library=libqti-perfd-client.so
-ro.vendor.product.brand=google
+ro.vendor.product.brand=Android
 ro.vendor.product.device=sailfish
-ro.vendor.product.manufacturer=Google
-ro.vendor.product.model=Pixel
-ro.vendor.product.name=sailfish
+ro.vendor.product.manufacturer=google
+ro.vendor.product.model=AOSP on msm8996
+ro.vendor.product.name=aosp_sailfish
 ro.vendor.vndk.version=26.1.0
 ro.wallpapers_loc_request_suw=true
 sdm.debug.disable_rotator_split=1
-setupwizard.theme=glif_v2_light
 telephony.lteOnCdmaDevice=1
 vidc.debug.perf.mode=2
```

I'm mostly concerned about `ro.control_privapp_permissions`. Will investigate further as soon as I've finished indexing the entire `8.1.0_r1` codebase.

anestisb, 2017-12-08:
The following two commits shed some light:
* [Control privileged permissions for priv-apps](https://android.googlesource.com/platform/frameworks/base/+/964d2ebb94c41d2cf1fb45b534e86c10e0ced3ac)
* [Support disabled/log/enforce in ro.control_privapp_permissions](https://android.googlesource.com/platform/frameworks/base/+/d072d1415452abffd55a5742e3fef5abdaadf791)

Seems that `ro.control_privapp_permissions` set to enforce is required to pass CTS. Will manually add for affected devices and see if something breaks. cc @thestinger

thestinger, 2017-12-08:
I'm already setting that myself but it required fixes.

thestinger, 2017-12-08:
I needed these changes for Android 8.0:

* https://github.com/CopperheadOS/platform_frameworks_base/commit/4b24b08586a89cb11d19939feb641afcc3840a0c
* https://github.com/CopperheadOS/platform_frameworks_base/commit/51575e87659f6e60030440bfe4e2e6f7c54ae213
* https://github.com/CopperheadOS/platform_frameworks_base/commit/cd084dc471db37519473217d3110eb4f7a962bbe
* https://github.com/CopperheadOS/platform_frameworks_base/commit/d2021bc263c6c955bb5683e9827e526bdfeebff0

I need to check and make sure it's all okay for Android 8.1 (mr1). I forgot to check this properly.

thestinger, 2017-12-08:
Documentation: https://source.android.com/devices/tech/config/perms-whitelist

thestinger, 2017-12-08:
I think my changes are still up-to-date.

thestinger, 2017-12-08:
Also note that the CTS does not yet require setting enforce to pass, but Nexus and Pixel devices do set it. It requires correct whitelists and presumably they'll make using enforce into a requirement for Android 9.0. The documentation covers that too: only correct whitelists required to pass, so at the moment AOSP doesn't pass...

anestisb, 2017-12-09:
Thanks @thestinger  for the info. Closing the issue since current state appears to be ok.

closed 2017-12-09 by anestisb