anestisb/android-prepare-vendor #113: Nexus Player support

kyvaith, 2017-12-14:
Hi,

Could You please extend your support for Nexus Player (figure)?

Thanks.

anestisb, 2017-12-14:
I don't have a Nexus Player device or have worked at all with this device. I'm guessing despite the different CPU arch, the configs approach should be identical.

I don't plan to provide support for this device. If you want you can start drafting the template configs and I can help along the process to iron them out. However, I don't have the capacity at the moment to do any kind of testing against the configs.

thestinger, 2018-01-05:
It's already end-of-life too, the last release was November 2017.

closed 2018-01-06 by anestisb

mydongistiny, 2018-02-26:
@kyvaith If you wanted to make a config you can start by basing it off of https://review.lineageos.org/#/c/207635/