anestisb/android-prepare-vendor #116: marlin - API27 Full Makefile naming

mydongistiny, 2017-12-31:
When generating marlin full config using latest source it makes a makefile named device-vendor-sailfish.mk instead of device-vendor-marlin.mk. Also the symlink names are now:

# Standalone symbolic links
PRODUCT_PACKAGES += \
    wcd9320_anc.bin__wcd9320_anc.bin__0 \
    wcd9320_mad_audio.bin__wcd9320_mad_audio.bin__1 \
    mbhc.bin__wcd9320_mbhc.bin__2 \
    toybox_vendor__acpi__3 \
    toybox_vendor__base64__4 \
    toybox_vendor__basename__5 \
    toybox_vendor__blockdev__6 \
    toybox_vendor__cal__7 \
    toybox_vendor__cat__8 \
    toybox_vendor__chcon__9 \
    toybox_vendor__chgrp__10 \
    toybox_vendor__chmod__11 \
    toybox_vendor__chown__12 \
    toybox_vendor__chroot__13 \
    toybox_vendor__chrt__14 \
    toybox_vendor__cksum__15 \
    toybox_vendor__clear__16 \

Not sure if the numbers are supposed to be there now as they weren't before.

https://hastebin.com/vuyuzulecu.pas  -  output of running execute-all
https://hastebin.com/lonihewalu.tex  - list of numbered symlinks

anestisb, 2018-01-01:
Thanks for reporting the issue. Will check the naming bug you mention. Most probably I regressed something when refactoring the configurations logic.

Regarding the numbering suffix check the following commits:
* https://github.com/anestisb/android-prepare-vendor/commit/edf65dfa0cd46b070fb4494cca4fdc46e600c2c5
* https://github.com/anestisb/android-prepare-vendor/commit/b060100ae9e521ee4cf7dc760eff441addb12bb4

The short story is that Pixel 2 factory images have a certain set of symbolic links where the basename of the src - dst pair is identical thus creating conflicts. So I've placed the number suffix counter as a simple workaround while also presenting a diff friendly logic with the old format.

anestisb, 2018-01-01:
Should be fixed now.

mydongistiny, 2018-01-02:
The naming problem is solved. The numbers are still there for some of the symlinks, but obviously  you know that. Thanks for the quick reply.

anestisb, 2018-01-02:
Yeah the number suffixes will remain for all generated symlink targets since it's required by the Pixel 2 devices to avoid conflicts.

closed 2018-01-02 by anestisb