anestisb/android-prepare-vendor #120: Implement factory zip caching

ypid, 2018-01-13:
Hey, thanks for your work.

I think it would be useful to not remove a successfully downloaded factory zip file if a later stage fails. I am downloading the zip now for the 4 time or so which takes enough time to allow me to open an issue :) What do you think? Possible issues for failing #114, `fuseext2` not installed, others.

anestisb, 2018-01-15:
If the script fails after the factory image file is downloaded, you don't have to download again. Just use the `-i|--img <path>` flag to point to the image file that has been downloaded from a previous execution. This doesn't apply if you directly set the output path to an AOSP root directory, since I don't want to pollute AOSP tree in case of errors.

Now the `--continue` flag for wget is indirectly automating the previous case if you set the same output path. The previous downloaded file is detected from wget and since it's downloaded properly it's not downloaded again. There was a similar PR (https://github.com/anestisb/android-prepare-vendor/pull/63), but I was a bit sceptical merging it because I wasn't sure if the continue flag was working for all wget implementations. If you're confident that it doesn't break across distros then it's fine for me to merge.

For the fuse issues I'm working on a more optimal solution. For now if the default debugfs fails in your setup, you will have to install fuse-ext2 and work with that.

ypid, 2018-01-20:
> If you're confident that it doesn't break across distros then it's fine for me to merge.

I am pretty confident that it works on all currently supported Debian and Ubuntu releases. I am not really speak about nor have I extensively tested this on other distros. But I think the risk should be pretty small compared to the benefits. wget runs in foreground and if there is something wrong with it, users will hopefully notice. As this was suggested already in #63 maybe we can give it a try?

closed 2018-01-20 by anestisb

captainhook, 2018-01-21:
Shouldn't there be a checksum verification to ensure the correct file has been downloaded?

ypid, 2018-01-22:
@captainhook Shouldn’t you open a new issue for this 😉 ? [Done](https://github.com/anestisb/android-prepare-vendor/issues/122). You are free to work on the issue as you raised the valid (in my eyes) concern.

commit d91da1e1 (2018-01-20): Implement factory zip caching/reusing

This causes wget to not re-download a perfectly happy, successfully
downloaded factory zip file if the script terminates at a later stage
and needs to be re-executed.

Closes: #120
PR #121 (2018-01-20): Implement factory zip caching/reusing
This causes wget to not re-download a perfectly happy, successfully downloaded factory zip file if the script terminates at a later stage and needs to be re-executed.

Closes: #120