anestisb/android-prepare-vendor #122: Ensure authenticity of downloaded zip file(s)

ypid, 2018-01-22:
As suggested by @captainhook in https://github.com/anestisb/android-prepare-vendor/issues/120#issuecomment-359247897:

From a security standpoint the authenticity of all downloaded file should be verified before further processing the files. Unfortunately, Google only provides SHA256 checksums. I would strongly prefer a digital signature over the files instead of a checksum.

As I don’t think we can get Google to sign their factory images zip files as a whole (using GnuPG for example), I guess this leaves us with three options:

1. Verify the signatures of the factory images itself after unpacking the zip file.
2. Maintain a database (text file checked into this repo) with all known good checksums for all downloaded files. (This might sound as nobody would so such a thing but that is not true. Refs: https://github.com/rootkovska/codehash.db, https://github.com/hamishcoleman/thinkpad-ec/blob/master/Descriptions.txt).
3. Ignore this issue. If the server gets hacked, TLS gets broken or Google gets evil then this would make targeting all/single users pretty easy.

#### Refs

* [Security challenges for the Qubes build process](https://blog.invisiblethings.org/2016/05/30/build-security.html)

thestinger, 2018-01-22:
For the Pixel 2 / Pixel 2 XL, Android Verified Boot 2.0 makes it easy to verify vbmeta, boot/dtbo and the system/vendor dm-verity hash trees. This is with CopperheadOS:

```
% /home/strcat/android/oreo-mr1-release/external/avb/avbtool verify_image --image vbmeta.img --key avb_pub.pem
Verifying image vbmeta.img using key at avb_pub.pem
vbmeta: Successfully verified SHA256_RSA2048 vbmeta struct in vbmeta.img
boot: Successfully verified sha256 hash of boot.img for image of 24408064 bytes
system: Successfully verified sha1 hashtree of system.img for image of 2641915904 bytes
vendor: Successfully verified sha1 hashtree of vendor.img for image of 515936256 bytes
dtbo: Successfully verified sha256 hash of dtbo.img for image of 2306048 bytes
```

The system/vendor images would still need to be verified via the hash trees, and that might be tricky without mounting them with dm-verity.

The bootloader/radio images are all signed but I don't have an easy way to verify those. On the positive side, they always get verified on the device.

anestisb, 2018-05-18:
I don't plan to add any logic to verify the integrity of the factory images that are downloaded from Google website. If any project users want to jump-in I'll be happy to support/merge their work.

closed 2018-08-11 by anestisb

ypid, 2020-01-25:
Just a hint, Hashbang OS has solved this by parsing the SHA256 checksums from Googles release pages and then pre-downloading them before calling android-prepare-vendor.
Ref: https://github.com/hashbang/aosp-build/blob/master/scripts/build-vendor

thestinger, 2020-01-26:
SHA256 hashes are not signatures and an attacker would just replace those alongside the zip files.

ypid, 2020-01-26:
You are right. I was not precise in my last comment. It is solved in the way I described it in my [initial comment](https://github.com/anestisb/android-prepare-vendor/issues/122#issue-290584065). Hashbang OS downloads the checksums once, and then everyone building a particular version will check for the same hash. Also, the hashes are signed by the Hashbang OS maintainers and can be easily re-parsed from the website and compared without downloading the files for every target. It is not as good as signatures of course but the best we can do it seems.