anestisb/android-prepare-vendor #124: ./execute-all.sh: ligne 642: VGEN_SCRIPT_EXTRA_ARGS[@]

thierryg, 2018-01-31:
Hello,

I wanted to repack the vendor for a google/sailfish 7.1.2 (NJH47F, Aug 2017) but even if the download are correct, the script have problem.

It stop by (see above for more detail)
./execute-all.sh: ligne 642: VGEN_SCRIPT_EXTRA_ARGS[@] : variable sans liaison

I just resynchronize with the current head of the git repo (my last sha1 is 9b2daac97436a76d90812f58a8689804cf5d0829 / Merge pull request #123 from NateDev100/patch-1)

Is it a known issue ?

Here is the detail : 

$ PATH=/home/tgayet/_aosp-git-repos/android-prepare-vendor/_tmp-jdk/jdk1.8.0_101/bin:$PATH ; ./execute-all.sh --device sailfish --buildID NJH47F --output $(pwd) --yes
[*] Setting output base to '/home/tgayet/_aosp-git-repos/android-prepare-vendor/sailfish/njh47f'

--{ Google Terms and Conditions
Downloading of the system image and use of the device software is subject to the
Google Terms of Service [1]. By continuing, you agree to the Google Terms of
Service [1] and Privacy Policy [2]. Your downloading of the system image and use
of the device software may also be subject to certain third-party terms of
service, which can be found in Settings > About phone > Legal information, or as
otherwise provided.

[1] https://www.google.com/intl/en/policies/terms/
[2] https://www.google.com/intl/en/policies/privacy/

[?] I have read and agree with the above terms and conditions - ACKNOWLEDGE [y|n]: yes
[*] Downloading image from 'https://dl.google.com/dl/android/aosp/sailfish-njh47f-factory-6fd9b2c4.zip'
--2018-01-31 17:01:01--  https://dl.google.com/dl/android/aosp/sailfish-njh47f-factory-6fd9b2c4.zip
Résolution de dl.google.com (dl.google.com)… 74.125.71.190, 74.125.71.136, 74.125.71.91, ...
Connexion à dl.google.com (dl.google.com)|74.125.71.190|:443… connecté.
requête HTTP transmise, en attente de la réponse… 200 OK
Taille : 1914506637 (1.8G) [application/zip]
Enregistre : «/home/tgayet/_aosp-git-repos/android-prepare-vendor/sailfish/njh47f/sailfish-njh47f-factory-6fd9b2c4.zip»

/home/tgayet/_aosp-git-repos/android-prepare-vendor/sailfish/n 100%[====================================================================================================================================================>]   1.78G  10.9MB/s    ds 3m 1s   

2018-01-31 17:04:02 (10.1 MB/s) - «/home/tgayet/_aosp-git-repos/android-prepare-vendor/sailfish/njh47f/sailfish-njh47f-factory-6fd9b2c4.zip» enregistré [1914506637/1914506637]

[*] Extracting '/home/tgayet/_aosp-git-repos/android-prepare-vendor/sailfish/njh47f/sailfish-njh47f-factory-6fd9b2c4.zip'
[*] Unzipping 'image-sailfish-njh47f.zip'
[*] Processing with 'API-25 naked' configuration
[*] '17' bytecode archive files will be repaired
[*] Repairing bytecode under /system partition using oatdump method
[!] '/framework/rcsservice.jar' not pre-optimized with sanity checks passed - copying without changes
[!] '/framework/embmslibrary.jar' not pre-optimized with sanity checks passed - copying without changes
[!] '/app/radioconfig/radioconfig.apk' not pre-optimized with sanity checks passed - copying without changes
[!] '/app/RCSBootstraputil/RCSBootstraputil.apk' not pre-optimized with sanity checks passed - copying without changes
[!] '/app/QAS_DVC_MSP/QAS_DVC_MSP.apk' not pre-optimized with sanity checks passed - copying without changes
[!] '/app/TimeService/TimeService.apk' not pre-optimized with sanity checks passed - copying without changes
[!] '/app/atfwd/atfwd.apk' not pre-optimized with sanity checks passed - copying without changes
[!] '/app/shutdownlistener/shutdownlistener.apk' not pre-optimized with sanity checks passed - copying without changes
[!] '/app/QtiTelephonyService/QtiTelephonyService.apk' not pre-optimized with sanity checks passed - copying without changes
[!] '/app/fastdormancy/fastdormancy.apk' not pre-optimized with sanity checks passed - copying without changes
[!] '/app/embms/embms.apk' not pre-optimized with sanity checks passed - copying without changes
[!] '/priv-app/CNEService/CNEService.apk' not pre-optimized with sanity checks passed - copying without changes
[!] '/priv-app/DiagMon/DiagMon.apk' not pre-optimized with sanity checks passed - copying without changes
[!] '/priv-app/CarrierEntitlement/CarrierEntitlement.apk' not pre-optimized with sanity checks passed - copying without changes
[!] '/priv-app/qcrilmsgtunnel/qcrilmsgtunnel.apk' not pre-optimized with sanity checks passed - copying without changes
[*] System partition successfully extracted & repaired at '/home/tgayet/_aosp-git-repos/android-prepare-vendor/sailfish/njh47f/factory_imgs_repaired_data'
./execute-all.sh: ligne 642: VGEN_SCRIPT_EXTRA_ARGS[@] : variable sans liaison

Some step haven't been started, so my blob haven't been generated :  

[*] Generating blobs for vendor/google_devices/sailfish
[*] Copying radio files '/home/tgayet/_aosp-git-repos/android-prepare-vendor/sailfish/nmf26v/vendor/google_devices/sailfish'
[*] Copying product files & generating 'sailfish-vendor-blobs.mk' makefile
[*] Generating 'device-vendor-sailfish.mk'
[*] Generating 'AndroidBoardVendor.mk'
  [*] Bootloader:8996-012001-1611091517
  [*] Baseband:8996-012511-1611190200
[*] Generating 'BoardConfigVendor.mk'
[*] Generating 'vendor-board-info.txt'
[*] Generating 'Android.mk'
[*] Gathering data from 'vendor/app' APK/JAR pre-builts
[*] Gathering data from 'vendor/framework' APK/JAR pre-builts
[*] Gathering data from 'proprietary/framework' APK/JAR pre-builts
[*] Processing standalone symlinks
[*] Generating signatures file
[*] All actions completed successfully
[*] Import '/home/tgayet/_aosp-git-repos/android-prepare-vendor/sailfish/nmf26v/vendor' to AOSP root

anestisb, 2018-02-01:
I cannot reproduce your issue. Can you please double check you git repo status (`git status` or `git diff`) for any manual edits you've might forgotten.

Also I'm not sure what `variable sans liaison` means. Can you please switch your bash locale (or translate to the matching error code)?

thierryg, 2018-02-01:
Hi, 
Thanks for your reply.
The string "variable sans liaison" means : "variable without link"
My git log say : 
$ git log
commit 9b2daac97436a76d90812f58a8689804cf5d0829
Merge: d99ae29 54ae863
Author: Anestis Bechtsoudis <anestisb@users.noreply.github.com>
Date:   Wed Jan 24 09:37:17 2018 +0400

    Merge pull request #123 from NateDev100/patch-1
    
    Fix typo

commit 54ae863bd2ac5365634f0682a43e4050b67a5fb4
Author: NateDev100 <31457698+NateDev100@users.noreply.github.com>
Date:   Tue Jan 23 18:11:46 2018 -0500

    Fix typo

commit d99ae29a9bcc4ccec323fc0cebb3a64cc0eb41a4
Merge: 9aac652 d91da1e
Author: Anestis Bechtsoudis <anestisb@users.noreply.github.com>
Date:   Sat Jan 20 18:32:25 2018 +0400

    Merge pull request #121 from ypid/use_wget-c
    
    Implement factory zip caching/reusing

anestisb, 2018-02-02:
Ok bug should be fixed in https://github.com/anestisb/android-prepare-vendor/commit/15f61c4a798ad98dbabdec287033264dea3cfe35. Could you please git pull and check if it works for your env?

closed 2018-02-03 by anestisb