anestisb/android-prepare-vendor #128: Failed to extract API level from build.prop

mikalackis, 2018-08-13:
I'm having problems extracting vendor from latest angler image (https://dl.google.com/dl/android/aosp/angler-opm6.171019.030.h1-factory-00138df6.zip)

Here is the output I get when running extract-all script (my repo is fully up-to-date):
`
mikalackis@mikalackis-X550LB:~/Projects/mikalackis/android-prepare-vendor$ ./execute-all.sh -d angler -a angler -b OPM6.171019.030.H1 -o nexus6p/
[*] Setting output base to '/home/mikalackis/Projects/mikalackis/android-prepare-vendor/nexus6p/angler/opm6.171019.030.h1'
 
--{ Google Terms and Conditions
Downloading of the system image and use of the device software is subject to the
Google Terms of Service [1]. By continuing, you agree to the Google Terms of
Service [1] and Privacy Policy [2]. Your downloading of the system image and use
of the device software may also be subject to certain third-party terms of
service, which can be found in Settings > About phone > Legal information, or as
otherwise provided.
 
[1] https://www.google.com/intl/en/policies/terms/
[2] https://www.google.com/intl/en/policies/privacy/
 
[?] I have read and agree with the above terms and conditions - ACKNOWLEDGE [y|n]: y
[*] Downloading image from 'https://dl.google.com/dl/android/aosp/angler-opm6.171019.030.h1-factory-00138df6.zip'
--2018-08-13 11:47:13--  https://dl.google.com/dl/android/aosp/angler-opm6.171019.030.h1-factory-00138df6.zip
Resolving dl.google.com (dl.google.com)... 216.58.209.174, 2a00:1450:400d:809::200e
Connecting to dl.google.com (dl.google.com)|216.58.209.174|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1064343610 (1015M) [application/zip]
Saving to: ‘/home/mikalackis/Projects/mikalackis/android-prepare-vendor/nexus6p/angler/opm6.171019.030.h1/angler-opm6.171019.030.h1-factory-00138df6.zip’
 
/home/mikalackis/Projects/mikalackis/android-prepar 100%[===================================================================================================================>]   1015M  1,07MB/s    in 17m 2s  
 
2018-08-13 12:04:15 (1017 KB/s) - ‘/home/mikalackis/Projects/mikalackis/android-prepare-vendor/nexus6p/angler/opm6.171019.030.h1/angler-opm6.171019.030.h1-factory-00138df6.zip’ saved [1064343610/1064343610]
 
[*] Extracting '/home/mikalackis/Projects/mikalackis/android-prepare-vendor/nexus6p/angler/opm6.171019.030.h1/angler-opm6.171019.030.h1-factory-00138df6.zip'
[*] Unzipping 'image-angler-opm6.171019.030.h1.zip'
[-] Failed to extract API level from build.prop
`

https://pastebin.com/vys9LRiY

anestisb, 2018-08-13:
Damn it another fuse related bug (Ubuntu 16.04.5 box). I'm so frustrated with the poor quality of Linux ext4 fuse tools.

Seems that in the case of this angler image ext4fuse (current default extract method for Linux) screws-up with some inode numbers, so the build.prop file is effectively an oat file.

```
$ mount | grep angler
ext4fuse on /home/anestisb/Developer/android/android-prepare-vendor/angler/opm6.171019.030.h1/factory_imgs_data/system type fuse.ext4fuse (ro,nosuid,nodev,relatime,user_id=1000,group_id=1000)
ext4fuse on /home/anestisb/Developer/android/android-prepare-vendor/angler/opm6.171019.030.h1/factory_imgs_data/vendor type fuse.ext4fuse (ro,nosuid,nodev,relatime,user_id=1000,group_id=1000)

$ file /home/anestisb/Developer/android/android-prepare-vendor/angler/opm6.171019.030.h1/factory_imgs_data/system/build.prop
/home/anestisb/Developer/android/android-prepare-vendor/angler/opm6.171019.030.h1/factory_imgs_data/system/build.prop: ELF 64-bit LSB shared object, ARM aarch64, version 1 (GNU/Linux), dynamically linked, BuildID[sha1]=a3ec4e62bfc1669e8b214f8427e090b3a509fcb9, stripped

$ strings /home/anestisb/Developer/android/android-prepare-vendor/angler/opm6.171019.030.h1/factory_imgs_data/system/build.prop | head -10
classpath
compiler-filter
quicken
concurrent-copying
true
debuggable
false
dex2oat-cmdline
--runtime-arg -Xms64m --runtime-arg -Xmx512m --class-loader-context=& --boot-image=out/target/product/angler/dex_bootjars/system/framework/boot.art --dex-file=out/target/common/obj/JAVA_LIBRARIES/com.android.mediadrm.signer_intermediates/javalib.jar --dex-location=/system/framework/com.android.mediadrm.signer.jar --oat-file=out/target/product/angler/obj/JAVA_LIBRARIES/com.android.mediadrm.signer_intermediates/oat/arm64/javalib.odex --android-root=out/target/product/angler/system --instruction-set=arm64 --instruction-set-variant=cortex-a53 --instruction-set-features=default --runtime-arg -Xnorelocate --compile-pic --no-generate-debug-info --generate-build-id --abort-on-hard-verifier-error --force-determinism --no-inline-from=core-oj.jar --compiler-filter=quicken
dex2oat-host
```

debugfs, fuse-ext2 & debugfs alternative method appear to work for this image. If I have time I'll try to isolate the bug and report it to upstream ext4fuse project.

mikalackis, 2018-08-13:
Thanks, confirming it works with fuse-ext2 so I'll stick with that for the time being.

stealthybox, 2019-06-03:
I believe I'm running into this on Ubuntu /w bullhead:
```
./execute-all.sh -d bullhead -f -b OPM7.181205.001 -o vendor
[*] Setting output base to '/home/stealthybox/Repos/android-prepare-vendor/android-prepare-vendor/vendor/bullhead/opm7.181205.001'

--{ Google Terms and Conditions
Downloading of the system image and use of the device software is subject to the
Google Terms of Service [1]. By continuing, you agree to the Google Terms of
Service [1] and Privacy Policy [2]. Your downloading of the system image and use
of the device software may also be subject to certain third-party terms of
service, which can be found in Settings > About phone > Legal information, or as
otherwise provided.

[1] https://www.google.com/intl/en/policies/terms/
[2] https://www.google.com/intl/en/policies/privacy/

[?] I have read and agree with the above terms and conditions - ACKNOWLEDGE [y|n]: y
[*] Downloading image from 'https://dl.google.com/dl/android/aosp/bullhead-opm7.181205.001-factory-5f189d84.zip'
--2019-06-03 15:10:47--  https://dl.google.com/dl/android/aosp/bullhead-opm7.181205.001-factory-5f189d84.zip
Resolving dl.google.com (dl.google.com)... 216.58.194.174, 2607:f8b0:4005:804::200e
Connecting to dl.google.com (dl.google.com)|216.58.194.174|:443... connected.
HTTP request sent, awaiting response... 416 Requested range not satisfiable

    The file is already fully retrieved; nothing to do.

[*] Using ext4fuse for image mounts
[*] Extracting '/home/stealthybox/Repos/android-prepare-vendor/android-prepare-vendor/vendor/bullhead/opm7.181205.001/bullhead-opm7.181205.001-factory-5f189d84.zip'
[*] Unzipping 'image-bullhead-opm7.181205.001.zip'
[-] Failed to extract API level from build.prop
```

```
cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=19.04
DISTRIB_CODENAME=disco
DISTRIB_DESCRIPTION="Pop!_OS 19.04"

uname -sprv
Linux 5.0.0-13-generic #14-Ubuntu SMP Mon Apr 15 14:59:14 UTC 2019 x86_64
```

stealthybox, 2019-06-03:
fuse-ext2 fails to mount for some reason:
```
[*] Using fuse-ext2 for image mounts
[*] Extracting '/home/stealthybox/Repos/android-prepare-vendor/android-prepare-vendor/vendor/bullhead/opm7.181205.001/bullhead-opm7.181205.001-factory-5f189d84.zip'
[*] Unzipping 'image-bullhead-opm7.181205.001.zip'
[-] '/tmp/android_img_extract.Aklheo/bullhead-opm7.181205.001-factory-5f189d84/images/system.img.raw' mount point missing indicates fuse mount error
[-] Factory images data extract failed
fusermount: entry for /home/stealthybox/Repos/android-prepare-vendor/android-prepare-vendor/vendor/bullhead/opm7.181205.001/factory_imgs_data/system not found in /etc/mtab
[-] '/home/stealthybox/Repos/android-prepare-vendor/android-prepare-vendor/vendor/bullhead/opm7.181205.001/factory_imgs_data/system' unmount failed
```

debugfs appears to be working :)