anestisb/android-prepare-vendor #130: PowerAnomaly frameworks

anestisb, 2018-08-15:
The following framework Jar are present in Android Pie sailfish images, although are not present in AOSP.

* system/framework/PowerAnomalyDataModemInterface.jar
* system/framework/PowerAnomalyStub.jar

So far all missing framework Jar were inserted in naked config. Adding them is postponed until the CDEX issue is resolved.

anestisb, 2018-08-15:
Same for "system/product/app/SSRestartDetector/SSRestartDetector.apk"

anestisb, 2018-09-02:
The oatdump [patch](https://gist.github.com/anestisb/26ecf8ae13746dc476eddd8d04a5dd23) and the standalone [utility](https://gist.github.com/anestisb/30265097ad9a5ea2f0ddf7e36db3f07d) are stable enough for testing. I'll probably proceed with the oatdump approach since I'm already compiling for the distribution.

Testing is scheduled for the next couple of days.

dan-v, 2018-09-02:
Thanks for the update @anestisb

anestisb, 2018-09-04:
Cdex support has been added via an oatdump patch. Latest revision of the patch is available [here](https://gist.github.com/anestisb/26ecf8ae13746dc476eddd8d04a5dd23). Will upstream the code to AOSP so that I don't need to maintain it independently.

anestisb, 2018-09-04:
Missing bytecode files have been added to the corresponding device configs.

closed 2018-09-04 by anestisb