anestisb/android-prepare-vendor #134: Certain device-version pairs give non booting builds

chirayudesai, 2018-10-04:
I am building for the Pixel devices, and am facing a peculiar issue.
The environment is plain AOSP, as a well as a slightly modified fork (same results with both)
Devices: Pixel 2 XL (taimen), Pixel XL (marlin)

Booting builds:
* marlin september ppr2.180905.006

Non-booting builds
* taimen september ppr2.180905.005
* marlin october ppr2.181005.003 (untested with plain AOSP yet) - and there's not that many changes from the working september vendor.
* taimen october ppr2.181005.003 (untested with plain AOSP yet)

The non-booting builds are stuck on the default "android" bootanimation, and there's no ADB access either to get logs.
The factory image extraction process didn't print any errors apart from being able to extract "/etc/fs_config_dirs". I'm using debugfs.

I'm not sure what's causing this, it could be my environment, however the builds work just fine if I swap out the vendor with the binaries Google provides. I'm waiting on something to test on a different environment.

All builds are built using 'make -j16', and flashed using 'fastboot -w flashall'

Any help / pointers would be appreciated. Thanks!

anestisb, 2018-10-15:
That is indeed a very strange behaviour. I'll try a fresh marlin ppr2.181005.003 and see if I can reproduce the problem. Unfortunately, if you don't have any logs or serial cable to get some data, out it would be really hard to debug the issue.

I would suggest that you clean your build output, check that src tree doesn't have any stashed/dirty patches, build a dist package (`make dist`), flash bootloader & radio, and then flash the entire update package (`fastboot -w update <aosp_img>.zip`).

chirayudesai, 2018-10-22:
The suggestion worked, and I think I _might_ know what the issue might be.

The `dist` build failed on the first try (unable to find aboot.img), and I diagnosed that to having the `vendor/google_devices/marlin` folder as a symlink (I did that in Oreo too, worked just fine back then) - Replacing that with a full copy of the contents seems to have worked, and I have a working marlin ppr2.181005.003 build.

Got the custom builds booting too, even taimen.

I guess that constitutes an error in my setup, sorry for the bother.

closed 2018-10-22 by chirayudesai