anestisb/android-prepare-vendor #141: blueline (Pixel 3) ends up with board=crosshatch in android-info.txt

thestinger, 2018-12-22:
This prevents the generated flashing factory images since it checks this when flashing the image zip. It's happening because there's only a shared BoardConfigVendor.mk using vendor/google_devices/crosshatch/vendor-board-info.txt as the source of information. Nothing uses the generated vendor/google_devices/blueline/vendor-board-info.txt.

anestisb, 2018-12-23:
@thestinger thanks for reporting. Unfortunately I didn't secure any free time yet to properly look the Pixel 3 support (initial support was contributed by other folks). Hopefully, will catch up with the backlog these days.

anestisb, 2018-12-30:
The BoardConfigVendor issue has been solved by creating two partials and wrap-around includes based on target.

However, I still can't solve the device-vendor.mk collision (see https://github.com/anestisb/android-prepare-vendor/commit/45c96466ca957e8068a37332f2eecb39368175cd). Let's hope that indeed the required blobs are identical, otherwise I'm not sure it's possible to solve without changes in AOSP.

closed 2019-01-09 by thestinger