anestisb/android-prepare-vendor #144: Crosshatch device testing

anestisb, 2018-12-30:
Test a clean build against a Pixel 3 XL device after commit https://github.com/anestisb/android-prepare-vendor/commit/d79c45d662b6b7b26bf01eb384241a7494c10e66 (various config areas & missing blobs were modified).

Unfortunately I wasn't able to secure yet a Pixel 3 XL device, so it would be great if some of the tool users can test this.

dan-v, 2019-01-19:
wow - @iMouath that's a very detailed report! I have a Pixel 3 XL using android-prepare-vendor (quite a few commits behind https://github.com/RattlesnakeOS/android-prepare-vendor but no changes) but it results in working audio and wifi. I haven't tried pulling in latest commits from android-prepare-vendor and testing though. Might be worth testing back on the same commit that I'm on.

iMouath, 2019-01-19:
It's very likely that I botched the build.

I'll try and read more about the build process, I'm confused about the https://developers.google.com/android/drivers I didn't use them which could explain why I'm missing hardware functionality.

anestisb, 2019-01-20:
@iMouath really appreciate the very detailed report. Will start working on them.

For WiFi & Audio I've created separate tasks to track them (see https://github.com/anestisb/android-prepare-vendor/issues/150 & https://github.com/anestisb/android-prepare-vendor/issues/151).

eSIM should be available if you use the full config:
https://github.com/anestisb/android-prepare-vendor/blob/7dee0eaf3ad65bb34be68fe6fe2922af122990ea/crosshatch/config.json#L230

For the assistance thing I'm not sure where it's implemented and if its something can be made available without the Google App blobs. Will investigate, with low priority though.

iMouath, 2019-01-21:
@anestisb I'm glad the report was useful to you.

I'll build another image using the full config when I get a chance and give you feedback on it.

regarding the assistance squeeze thingy (Active Edge) I suspect it's packaged under **SystemUIGoogle.apk** but as you said it isn't a core feature and a blob so not really a concern at all. and it's reported to be broken in all AOSP roms.