anestisb/android-prepare-vendor #147: Taimen: Vendor generation failed

M83tUt3, 2019-01-04:
This happens on my Fedora install but I'm getting the same results using an Ubuntu 18.04 live CD. It is working fine using the 0.5.0 tag.

fedora@fedora:/home/fedora/android-prepare-vendor$ ./execute-all.sh --debugfs --yes --device taimen --buildID PQ1A.181205.002 --output out/
[*] Setting output base to '/home/fedora/android-prepare-vendor/out/taimen/pq1a.181205.002'

--{ Google Terms and Conditions
Downloading of the system image and use of the device software is subject to the
Google Terms of Service [1]. By continuing, you agree to the Google Terms of
Service [1] and Privacy Policy [2]. Your downloading of the system image and use
of the device software may also be subject to certain third-party terms of
service, which can be found in Settings > About phone > Legal information, or as
otherwise provided.

[1] https://www.google.com/intl/en/policies/terms/
[2] https://www.google.com/intl/en/policies/privacy/

[?] I have read and agree with the above terms and conditions - ACKNOWLEDGE [y|n]: yes
[*] Downloading image from 'https://dl.google.com/dl/android/aosp/taimen-pq1a.181205.002-factory-4d01cf61.zip'
--2019-01-04 19:09:36--  https://dl.google.com/dl/android/aosp/taimen-pq1a.181205.002-factory-4d01cf61.zip
Resolving dl.google.com (dl.google.com)... 172.217.20.78, 2a00:1450:400e:80a::200e
Connecting to dl.google.com (dl.google.com)|172.217.20.78|:443... connected.
HTTP request sent, awaiting response... 416 Requested range not satisfiable

    The file is already fully retrieved; nothing to do.

[*] Using debugfs for image mounts
[*] Extracting '/home/fedora/android-prepare-vendor/out/taimen/pq1a.181205.002/taimen-pq1a.181205.002-factory-4d01cf61.zip'
[*] Unzipping 'image-taimen-pq1a.181205.002.zip'
[*] Processing with 'API-28 naked' configuration
[*] '14' bytecode archive files will be repaired
[*] Repairing bytecode under /system partition using oatdump method
[!] '/framework/embmslibrary.jar' bytecode is not stripped - copying without changes
[!] '/framework/qcrilhook.jar' bytecode is not stripped - copying without changes
[!] '/framework/com.quicinc.cne.api-V1.0-java.jar' bytecode is not stripped - copying without changes
[!] '/framework/rcsimssettings.jar' bytecode is not stripped - copying without changes
[!] '/framework/cneapiclient.jar' bytecode is not stripped - copying without changes
[!] '/framework/com.quicinc.cne.jar' bytecode is not stripped - copying without changes
[!] '/app/uceShimService/uceShimService.apk' bytecode is not stripped - copying without changes
[!] '/app/ims/ims.apk' bytecode is not stripped - copying without changes
[!] '/app/embms/embms.apk' bytecode is not stripped - copying without changes
[!] '/priv-app/qcrilmsgtunnel/qcrilmsgtunnel.apk' bytecode is not stripped - copying without changes
[!] '/priv-app/CNEService/CNEService.apk' bytecode is not stripped - copying without changes
cp: cannot open '/home/fedora/android-prepare-vendor/out/taimen/pq1a.181205.002/factory_imgs_data/system/system/etc/fs_config_dirs' for reading: Permission denied
[!] Failed to copy '/etc/fs_config_dirs' - skipping
[*] System partition successfully extracted & repaired at '/home/fedora/android-prepare-vendor/out/taimen/pq1a.181205.002/factory_imgs_repaired_data'
[*] Generating 'taimen' vendor blobs
[*] Copying radio files '/home/fedora/android-prepare-vendor/out/taimen/pq1a.181205.002/vendor/google_devices/taimen'
[*] Copying product files & generating 'taimen-vendor-blobs.mk' makefile
[*] Generating 'device-vendor.mk'
[*] Generating 'AndroidBoardVendor.mk'
  [*] Bootloader:TMZ20k
  [*] Baseband:g8998-00253-1809191716
[*] Generating 'BoardConfigVendor.mk'
[-] Vendor generation failed

anestisb, 2019-01-05:
Fixed in https://github.com/anestisb/android-prepare-vendor/commit/697416b663ec5cf04b5ad4dc36e6de90dfb7b797

M83tUt3, 2019-01-05:
Indeed, much thanks.

closed 2019-01-05 by M83tUt3