anestisb/android-prepare-vendor #149: dtbo shouldn't be included by android-prepare-vendor

thestinger, 2019-01-09:
This is already included by the AOSP build setup and android-prepare-vendor is treating it as a radio file which breaks applying update packages.

anestisb, 2019-01-09:
Fixed in https://github.com/anestisb/android-prepare-vendor/commit/a2a5face3ea050c76ea743381d71e3bf14c7c1c3. However, I found it was missing when building a dist package. Let me try a fresh build to verify with current master.

anestisb, 2019-01-09:
Ok verified dtbo.img is part of the OTA image. Some must have been messed up in my previous builds.

```
$ zipinfo out_userdebug_release_Linux_9.0.0_r21/aosp_prod/dist/aosp_blueline-img-eng.anestisb.zip dtbo.img
-rw----     2.0 fat  8388608 bl defN 08-Jan-01 00:00 dtbo.img
```

thestinger, 2019-01-16:
Seems resolved.

closed 2019-01-16 by thestinger