anestisb/android-prepare-vendor #150: WiFi not working in crosshatch

anestisb, 2019-01-20:
See bug report & logs in https://github.com/anestisb/android-prepare-vendor/issues/144#issuecomment-455820354

anestisb, 2019-01-20:
Related entries from logcat:

```
01-18 07:24:50.795  1122  6907 D WifiService: setWifiEnabled: true pid=3318, uid=10022, package=com.android.systemui
01-18 07:24:53.006   810   810 E android.hardware.wifi@1.0-service: Failed to write driver state control param: Invalid argument
01-18 07:24:53.006   810   810 E android.hardware.wifi@1.0-service: Failed to load WiFi driver
01-18 07:24:53.006   810   810 E android.hardware.wifi@1.0-service: Failed to initialize firmware mode controller
01-18 07:24:53.007   810   810 E android.hardware.wifi@1.0-service: Wifi HAL start failed
```

thestinger, 2019-04-02:
This works fine for me.