anestisb/android-prepare-vendor #151: Audio not working in crosshatch

anestisb, 2019-01-20:
See bug report & logs in https://github.com/anestisb/android-prepare-vendor/issues/144#issuecomment-455820354

anestisb, 2019-01-20:
Related entries from logcat:

```
01-18 07:25:40.047   823  3176 W APM_AudioPolicyManager: getOutputForDevice() could not find output for stream 1, sampling rate 48000, format 0x1, channels 0x3, flags 0x4
01-18 07:25:40.047   823  3176 E AudioFlinger: createTrack() getOutputForAttr() return error -38 or invalid output handle
01-18 07:25:40.048  1122  2861 E IAudioFlinger: createTrack returned error -38
01-18 07:25:40.048  1122  2861 E AudioTrack: AudioFlinger could not create track, status: -38 output 281594344
01-18 07:25:40.048  1122  2861 E SoundPool: Error creating AudioTrack
01-18 07:25:40.048  1122  2861 D AudioTrack: no metrics gathered, track status=-38
```

thestinger, 2019-04-02:
This works fine for me.

jourd, 2019-04-05:
I got this working by using debugfs instead of ext4fuse (ubuntu 18.04).
I get "Failed to copy '/etc/fs_config_dirs' - skipping" mentioned in another issue but audio is working fine. Fair trade.