anestisb/android-prepare-vendor #152: oatdump dependencies download failed

yangtianrui95, 2019-02-01:
Hello
I use execute-all script to download vendor,  but it show this message "oatdump dependencies download failed", how to solve it ?

Here is my usage.

```shell
bash5-3.2$ ./execute-all.sh -d sailfish -b OPR3.170623.008 -k -o $(pwd)
[*] Setting output base to '/Volumes/android/android-prepare-vendor/sailfish/opr3.170623.008'

--{ Google Terms and Conditions
Downloading of the system image and use of the device software is subject to the
Google Terms of Service [1]. By continuing, you agree to the Google Terms of
Service [1] and Privacy Policy [2]. Your downloading of the system image and use
of the device software may also be subject to certain third-party terms of
service, which can be found in Settings > About phone > Legal information, or as
otherwise provided.

[1] https://www.google.com/intl/en/policies/terms/
[2] https://www.google.com/intl/en/policies/privacy/

[?] I have read and agree with the above terms and conditions - ACKNOWLEDGE [y|n]: y
[*] Downloading image from 'https://dl.google.com/dl/android/aosp/sailfish-opr3.170623.008-factory-bb8bb3a3.zip'
--2019-02-01 16:25:06--  https://dl.google.com/dl/android/aosp/sailfish-opr3.170623.008-factory-bb8bb3a3.zip
Resolving dl.google.com (dl.google.com)... 172.217.161.174
Connecting to dl.google.com (dl.google.com)|172.217.161.174|:443... connected.
HTTP request sent, awaiting response... 416 Requested range not satisfiable

    The file is already fully retrieved; nothing to do.

[*] Using ext4fuse for image mounts
[*] Extracting '/Volumes/android/android-prepare-vendor/sailfish/opr3.170623.008/sailfish-opr3.170623.008-factory-bb8bb3a3.zip'
[*] Unzipping 'image-sailfish-opr3.170623.008.zip'
[*] Processing with 'API-26 naked' configuration
[*] First run detected - downloading oatdump host bin & lib dependencies
--2019-02-01 16:26:14--  https://onedrive.live.com/download?cid=D1FAC8CC6BE2C2B0&resid=D1FAC8CC6BE2C2B0!561&authkey=ABu-oqJbQDwQ-ZQ
Resolving onedrive.live.com (onedrive.live.com)... 64.233.179.118
Connecting to onedrive.live.com (onedrive.live.com)|64.233.179.118|:443... connected.
WARNING: no certificate subject alternative name matches
	requested host name 'onedrive.live.com'.
HTTP request sent, awaiting response... 404 Not Found
2019-02-01 16:26:16 ERROR 404: Not Found.

--2019-02-01 16:26:16--  https://onedrive.live.com/download?cid=D1FAC8CC6BE2C2B0&resid=D1FAC8CC6BE2C2B0%21561&authkey=ABu-oqJbQDwQ-ZQ
Reusing existing connection to onedrive.live.com:443.
HTTP request sent, awaiting response... 404 Not Found
2019-02-01 16:26:16 ERROR 404: Not Found.

[-] oatdump dependencies download failed
```

anestisb, 2019-02-05:
Thanks for reporting. Based on the printed warning it seems to be some certificate issue with MSFTs OneDrive services. Double checking now if it was a temp issue or still present.

anestisb, 2019-02-05:
I can't reproduce from here. Can you double check if the issue still exists?

closed 2019-03-25 by anestisb