anestisb/android-prepare-vendor #153: Bytecode repair fails for a Google Nexus factory image

misterzed88, 2019-02-07:
Running version 0.5.0 or later of the tools (including the latest master), execute-all.sh fails on a factory image from Google:
https://dl.google.com/dl/android/aosp/angler-n2g48c-factory-6a21e528.zip

The reason is that bytecode repair with the oatdump method fails.

This must be a newly introduced bug since version 0.4.1 works fine.

The full output log is provided below.

$ sudo ./execute-all.sh -d angler -b n2g48c -o /mnt/HD/Android/vendor/ -i /mnt/HD/Android/images/angler-n2g48c-factory-6a21e528.zip
[*] Setting output base to '/mnt/HD/Android/vendor/angler/n2g48c'
[*] Running as root - using loopback for image mounts
[*] Extracting '/mnt/HD/Android/images/angler-n2g48c-factory-6a21e528.zip'
[*] Unzipping 'angler-n2g48c-factory-6a21e528.zip'
[*] Processing with 'API-25 full' configuration
[*] '18' bytecode archive files will be repaired
[*] Repairing bytecode under /system partition using oatdump method
[-] '/app/HwMMITest/HwMMITest.apk' DEX export failed
MAGIC:
oat
088

LOCATION:
/mnt/HD/Android/vendor/angler/n2g48c/factory_imgs_data/system/app/HwMMITest/oat/arm64/HwMMITest.odex

CHECKSUM:
0x60142334

INSTRUCTION SET:
Arm64

INSTRUCTION SET FEATURES:
smp,a53

DEX FILE COUNT:
1

EXECUTABLE OFFSET:
0x00038000

INTERPRETER TO INTERPRETER BRIDGE OFFSET:
0x00000000

INTERPRETER TO COMPILED CODE BRIDGE OFFSET:
0x00000000

JNI DLSYM LOOKUP OFFSET:
0x00000000

QUICK GENERIC JNI TRAMPOLINE OFFSET:
0x00000000

QUICK IMT CONFLICT TRAMPOLINE OFFSET:
0x00000000

QUICK RESOLUTION TRAMPOLINE OFFSET:
0x00000000

QUICK TO INTERPRETER BRIDGE OFFSET:
0x00000000

IMAGE PATCH DELTA:
0 (0x00000000)

IMAGE FILE LOCATION OAT CHECKSUM:
0x444fff23

IMAGE FILE LOCATION OAT BEGIN:
0x70dc4000

KEY VALUE STORE:
classpath = &
compiler-filter = speed
debuggable = false
dex2oat-cmdline = --runtime-arg -Xms64m --runtime-arg -Xmx512m --runtime-arg -classpath --runtime-arg & --boot-image=out/target/product/angler/dex_bootjars/system/framework/boot.art --dex-file=out/target/product/angler/obj/APPS/HwMMITest_intermediates/oat/arm64/package.odex.input --dex-location=/system/app/HwMMITest/HwMMITest.apk --oat-file=out/target/product/angler/obj/APPS/HwMMITest_intermediates/oat/arm64/package.odex --android-root=out/target/product/angler/system --instruction-set=arm64 --instruction-set-variant=cortex-a53 --instruction-set-features=default --include-patch-information --runtime-arg -Xnorelocate --no-generate-debug-info --abort-on-hard-verifier-error --no-inline-from=core-oj.jar --compile-pic
dex2oat-host = X86_64
has-patch-info = true
image-location = out/target/product/angler/dex_bootjars/system/framework/arm64/boot.art:out/target/product/angler/dex_bootjars/system/framework/arm64/boot-core-libart.art:out/target/product/angler/dex_bootjars/system/framework/arm64/boot-conscrypt.art:out/target/product/angler/dex_bootjars/system/framework/arm64/boot-okhttp.art:out/target/product/angler/dex_bootjars/system/framework/arm64/boot-core-junit.art:out/target/product/angler/dex_bootjars/system/framework/arm64/boot-bouncycastle.art:out/target/product/angler/dex_bootjars/system/framework/arm64/boot-ext.art:out/target/product/angler/dex_bootjars/system/framework/arm64/boot-framework.art:out/target/product/angler/dex_bootjars/system/framework/arm64/boot-telephony-common.art:out/target/product/angler/dex_bootjars/system/framework/arm64/boot-voip-common.art:out/target/product/angler/dex_bootjars/system/framework/arm64/boot-ims-common.art:out/target/product/angler/dex_bootjars/system/framework/arm64/boot-apache-xml.art:out/target/product/angler/dex_bootjars/system/framework/arm64/boot-org.apache.http.legacy.boot.art
native-debuggable = false
pic = true

SIZE:
461196

[-] System partition bytecode repair failed

anestisb, 2019-02-08:
@misterzed88 thanks for reporting. I'll try to reproduce and investigate. Most likely an update in the factory image apps is not compatible with oatdump decompilation utilities.

anestisb, 2019-02-10:
Should be fixed in https://github.com/anestisb/android-prepare-vendor/commit/cf8e900e216fc07c501f3f4c9c5427ff3ccc4943. Very weird though as to why it happens. The `--header-only` does not affect the paths that do the actual Dex export. Seems like an oatdump internal bug that is triggered if the Dex methods are not traversed.

misterzed88, 2019-02-18:
Nice that you found the cause and a quick workaround! (so that's why this problem surfaced after the 0.4.1 release: the optimization was introduced after that release). I assume it's not worth the effort troubleshooting and patching an old oatdump version, with the workaround in place, with only minor performance penalty.

Thank you for spending the time and the effort, making a great tool!

closed 2019-02-23 by anestisb