anestisb/android-prepare-vendor #159: sargo (Pixel 3a) support

FeeJai, 2019-05-08:


transitracer, 2019-05-12:
Is there anyone who can give a short summary of how the config files for existing devices have been created? As in, what steps would be needed to create one for sargo and bonito. I've got a sargo device in the mail and would be willing to put in the work if there was a general outline of any sort.

FeeJai, 2019-05-12:
+1. I also got a sargo in the mail, and would also be wiling to help

PieDelivery, 2019-05-13:
@dan-v pointed out [this commit](https://github.com/anestisb/android-prepare-vendor/pull/137/files) as the one that added support for the Pixel 3, if that's a place to start. I just got my Pixel 3a so I'd be willing to help.

PieDelivery, 2019-05-19:
I started a fork of android-prepare-vendor to mess around with (https://github.com/PieDelivery/android-prepare-vendor) and forked rattlesnakeos-stack to build it with. Not gonna lie I have no idea what I'm doing so any help or input would be appreciated. As of now, the image successfully builds but after flashing it the phone gets stuck in a boot loop.

PieDelivery, 2019-05-21:
Another update to this for those interested: the phone works fine when using the --full flag with the config file that I made, so it looks like I'm going to need to move some things around to get the naked build working. I'm going to open a few issues in my repo if you want to help. Just testing things and opening issues for what's wrong will help too.

anestisb, 2019-05-21:
@PieDelivery thank you for interest into committing to the project. Support for Pixel 3a devices is on the roadmap, although since I don't have any devices yet I can't really do anything from my side.

I would suggest that you start by copying across the Pixel 3 configuration and start working backwards the configs to make them compile with AOSP. Then you can continue with actual device flashing and testing.

PieDelivery, 2019-05-21:
@anestisb that's what I've done, as of now everything is working correctly for the most part with my fork, I just need to do a little more testing. Is there anything specifically that you recommend testing?

anestisb, 2019-05-22:
The usual suspects I start with are:

1. Fingerprint
2. Bluetooth
3. WiFi
4. 3G/4G
5. HotSpot
6. NFC
7. Sound
8. Place/receive calls/sms

ifbizo, 2019-05-24:
I get these messages in logcat, there might be more missing. I'm new to this and not sure of their significance though, I'm looking into it.

> 02-04 09:31:11.077   572   572 E QSEECOMD: : Init dlopen(libspl.so, RLTD_NOW) is failed.... dlopen failed: library "libspl.so" not found
> 05-23 04:25:40.359   741   741 E CHIUSECASE: chxutils.cpp:526 LibMap() Failed to load library libqti-perfd-client.so error dlopen failed: library "libqti-perfd-client.so" not found
> 05-23 04:25:45.351   977   977 D C2Store : could not dlopen libstagefright_soft_c2xaacdec.so: dlopen failed: library "libstagefright_soft_c2xaacdec.so" not found

anestisb, 2019-05-27:
@ifbizo the quotted error messages appear to be just log noise. None of the mentioned libraries appear to be present in factory images. I'm guessing some shared code across devices tries to load them across the border without actually needed all the time. Have you noticed any glitches in related functionalities (e.g. media codecs)?

PieDelivery, 2019-05-29:
@anestisb I noticed that "system/etc/permissions/com.google.modemservice.xml" exists in both the full and naked configs in the blueline and crosshatch repos, so I kept it in the sargo and bonito ones. Is there a reason for this or can the one in the full config be deleted?

anestisb, 2019-05-29:
@PieDelivery good catch. Fixed in https://github.com/anestisb/android-prepare-vendor/commit/344b96c5cb543875a7d95f6fcf6ff56a33deeea5.

FeeJai, 2019-06-03:
I have finally received my 3a. Already in love with it. Will have time to test @PieDelivery's config file on the weekend.

PieDelivery, 2019-06-03:
@FeeJai the first Pixel 3a update came out today, which combined the April, May, and June security updates. I won’t be able to look at it until this weekend, so let us know if there are any issues with the new update

anestisb, 2019-12-29:
Large done already. Not tested by me personally due to lack of 3a devices.

closed 2019-12-29 by anestisb