anestisb/android-prepare-vendor #161: eSIM manger crashes on sargo

PieDelivery, 2019-05-23:
eSIM manager constantly crashes when building with the full configuration on sargo. This does not affect normal SIM use, or the naked config.

Moved from: https://github.com/PieDelivery/android-prepare-vendor/issues/1

anestisb, 2019-05-24:
@PieDelivery can you paste relative crash logs from adb/serial?

Also can you give it a try with current head (noticed you missed some libs)?

PieDelivery, 2019-05-24:
Here's the entire log, in case you're interested in anything else. If you scroll down to the bottom you can see it crashing (at around line 6300). This is without the changes you made, I'm rebuilding it with those changes but it'll take a few hours.

[logcat.txt](https://github.com/anestisb/android-prepare-vendor/files/3217952/logcat.txt)

PieDelivery, 2019-05-24:
Flashed it with the changes, everything still seems to work, didn't seem to change anything with the eSIM, however. Here's the log, the errors are at the very end.

[logcat2.txt](https://github.com/anestisb/android-prepare-vendor/files/3218859/logcat2.txt)

anestisb, 2019-05-26:
Ok so the missing `com.google.android.gsf.gservices` content provider is actually a Google Service package. Therefore, the full image cannot be run if not bundled with GApps. I'm guessing you're trying the full config without them right?

PieDelivery, 2019-05-26:
Yes I'm tested the full image without GApps.

anestisb, 2019-05-27:
Well it seems that the eSIM packages cannot work without GApps. At any case the full config is meant to be used with GApps so usually I test it under the context of having both. Have you tried installing GApps on the top (or bundle them with the main AOSP build)?

PieDelivery, 2019-05-28:
That makes a lot of sense, I didn't know it was assumed to be used with GApps. I installed microG to test. The crashing stopped, and I was able to get to the screen for scanning a QR code for an eSIM unlike before (I don't have an actual eSIM to test). I'll mark this as closed.

closed 2019-05-28 by PieDelivery