anestisb/android-prepare-vendor #163: 'dexrepair' command not found

MichaB189, 2019-06-20:
Running ./scripts/system-img-repair.sh
results in 'dexrepair' command not found

dexrepair is located in both android-prepare-vendor/hostTools/Linux/bin and /Darwin/bin.
Furthermore I downloaded the dexrepair repo from github and ran the make.shell and I'm still getting the same error

What am I doing wrong? Or is this a fault? Thanks

PieDelivery, 2019-06-20:
The easiest way would probably be to add Linux/bin to your path (assuming you're on Linux) like `export PATH="$PATH:/path/to/bin"`.

You could also add a symlink in /usr/local/bin or something like that.

MichaB189, 2019-06-24:
> The easiest way would probably be to add Linux/bin to your path (assuming you're on Linux) like `export PATH="$PATH:/path/to/bin"`. Note that this won't be permanent and you'd need to rerun it every session.
> 
> edit: also, it's important to note that `execute-all.sh` takes care of this itself, so you'd only need to do this when running the individual scripts.

Thank you. I solved it by running the execute-all shell instead of the scripts seperately

closed 2019-07-21 by anestisb