anestisb/android-prepare-vendor #165: No valid OS found

MichaB189, 2019-06-26:
Im building clean AOSP with android-prepare-vendor

repo init -u https://android.googlesource.com/platform/manifest -b android-9.0.0_r40
repo sync

I remove AOSP/vendor/google_devices/*

PQ3A.190605.003 matches android-9.0.0_r40

sudo ./execute-all.sh -d blueline -b PQ3A.190605.003 -o out 

I copy the files generated:
from:
/android-prepare-vendor-master/out/blueline/pq3a.190605.003/vendor/google_devices/*
to: AOSP/vendor/google_devices

I then run make clean
make clobber
make -j12

cd AOSP/out/target/product/blueline
fastboot flashall -w

adb reboot bootloader
fastboot flashing lock

No Valid OS found
g.co/ABH

Am I missing something, either regarding copying the right files from android-prepare-vendor or something other?

PieDelivery, 2019-06-30:
Does it boot with the bootloader unlocked (i.e. don't run `fastboot flashing lock` after flashing the image and boot it)? You might need to flash the phone with the key you used to sign the image with `fastboot flash avb_custom_key key_name.bin` before locking it.

anestisb, 2019-12-29:
@MichaB189  do you still face the issue?

MichaB189, 2020-01-31:
Yes, it does boot with the bootloader unlocked

MichaB189, 2020-01-31:
I tried again now. WIth the August 2019 version. As I've heard there are some issues with Android 10

MichaB189, 2020-01-31:
> Does it boot with the bootloader unlocked (i.e. don't run `fastboot flashing lock` after flashing the image and boot it)? You might need to flash the phone with the key you used to sign the image with `fastboot flash avb_custom_key key_name.bin` before locking it.

Thanks!
I did not do that. Could you supply me with additional information regarding signing?
I had factory images from Google before flashing my vanilla AOSP build with android-prepare-vendor. Is there some in depth tutorial explaining the process of signing the keys, what keys I should sign and so forth? Thanks again!