anestisb/android-prepare-vendor #172: Nexus ToS accept request failed - Images download failed

MichaB189, 2019-09-30:
Running this command: 
sudo vendor/android-prepare-vendor/execute-all.sh -d blueline -b pie -o /some/dir

Results in this:
[*] Setting output base to '/some/dir/blueline/pie'

--{ Google Terms and Conditions
Downloading of the system image and use of the device software is subject to the
Google Terms of Service [1]. By continuing, you agree to the Google Terms of
Service [1] and Privacy Policy [2]. Your downloading of the system image and use
of the device software may also be subject to certain third-party terms of
service, which can be found in Settings > About phone > Legal information, or as
otherwise provided.

[1] https://www.google.com/intl/en/policies/terms/
[2] https://www.google.com/intl/en/policies/privacy/

[?] I have read and agree with the above terms and conditions - ACKNOWLEDGE [y|n]: yes

[-] Nexus ToS accept request failed
[-] Images download failed

agolden, 2019-10-05:
After futzing around with it, it seems like Google has gotten rid of the required csrf token.  All I had to do to get this to work was comment out the check for 200 in download-nexus-image.sh

```
if [[ "$response" != "200" ]]; then
  echo "[-] Nexus ToS accept request failed"
  abort 1
fi
```

anestisb, 2019-10-12:
Fixed from https://github.com/anestisb/android-prepare-vendor/pull/171. Thanks for reporting.

closed 2019-10-12 by anestisb