anestisb/android-prepare-vendor #176: download_nexus_image.sh error Image URL not found

carlgreen, 2020-05-15:
I am running download_nexus_image.sh similar to:
./download-nexus-image.sh -d bonito -b qq2a.200501.001.b2 -y -o /tmp/

The output (after the TOS) is:
[-] Image URL not found

Possibly the page has been changed to require clicking on a button prior to the real content loading? It seems to add a cookie.

I have been able to make this work by modifying the line following "# Then retrieve the index page" from:
url=$(curl -L -b "$COOKIE_FILE" --silent "$GURL" | \
to:
url=$(curl -L -b "$COOKIE_FILE" -b devsite_wall_acks=nexus-image-tos --silent "$GURL" | \

dan-v, 2020-05-15:
@carlgreen - there is a fix for this in our fork https://github.com/AOSPAlliance/android-prepare-vendor/commit/7a35f651486ab9fd5bec8935555ef549aa25b2c9. @anestisb in another issue has said he likely won't be able to support this project until June time frame.

carlgreen, 2020-05-15:
Oh cool, thanks. I couldn't find it mentioned in any issues.