anestisb/android-prepare-vendor #177: Stuck in bootloader

irmas5, 2020-11-01:
Hi. I hard bricked a pixel 2xl. i'm stuck in bootloader with no os and no recovery, when i locked back bootloader they wiped all system and recovery. Maybe here are some way how to fix? Thank you

thestinger, 2020-11-01:
You should ask on the #grapheneos IRC / Matrix channel. You should probably just unlock the device and flash again.

thestinger, 2020-11-01:
You should probably just unlock the device and flash again. You should also use android-prepare-vendor from AOSP Alliance which is being actively developed.

irmas5, 2020-11-01:
I can't flash anything becouse i don't have recovery, when i trying fastboot image file for a pixel 2xl saying failed flash in locked state.

thestinger, 2020-11-01:
Recovery isn't where you flash images. Unlock the bootloader.

irmas5, 2020-11-01:
How, if I can't flash in fastboot? Saying failed flash in a locked state

thestinger, 2020-11-01:
> Unlock the bootloader.

thestinger, 2020-11-01:
Not going to try helping further because you aren't listening.

irmas5, 2020-11-01:
Sorry, I'm really not a tech, and I don't know how to fix that. I', now without a phone.

thestinger, 2020-11-01:
The issue is that you aren't listening. Read what I wrote. No point in a conversation if you don't read what I write.

irmas5, 2020-11-01:
I reading. Unlock bootloader. Just how? Can you help me?

thestinger, 2020-11-01:
The opposite of locking it which is what you said you did at the start...

thestinger, 2020-11-01:
You need to do your own research. I don't think filing an issue here makes my sense. I suggest closing it and reading about this.

irmas5, 2020-11-01:
I searched on xda, Reddit. Nowhere found how to unlock bootloader from the fastboot menu.

thestinger, 2020-11-01:
The opposite of locking the bootloader: unlock instead of lock.

irmas5, 2020-11-01:
I tried, but showing ...
FAILED (remote: unknown command)
finished. total time: 0.006s
PS C:\adb> fastboot flashing unlock
...
FAILED (remote: Flashing Unlock is not allowed
)
finished. total time: 0.016s

irmas5, 2020-11-01:
PS C:\adb> fastboot flashing unlock_critical
...
FAILED (remote: Flashing Unlock is not allowed
)
finished. total time: 0.007s

thestinger, 2020-11-01:
Did you disable OEM unlocking in the OS?

irmas5, 2020-11-01:
I can't enter to os.
[image]
[image]

irmas5, 2020-11-01:
After I relocked the bootloader I think wiped all partition, os, and recovery.

irmas5, 2020-11-01:
I was on custom rom.

irmas5, 2020-11-01:
In oem was switched of.

closed 2020-11-02 by irmas5